project-chip/connectedhomeip #38980: [BUG] [TC-ACE-2.3] Test case failure but not having any test case step details in Test plan

ashley-se, 2025-05-15:
### Reproduction steps

1. This is new added test case in 1.5
2. This test is run in python docker follow the instruction/command in test plan with argument set timeout to 10000
3. I can see the DUT is commissioned
4. Test failure show at last
5. Try to understand the detail steps in the test case, but when visit the 'ACE' sheet in test plan, i don't see TC-ACE-2.3 listed there

Command:
python3 TC_AccessChecker.py  --commissioning-method ble-thread --thread-dataset-hex 0e08000000000001000035060004001fffe00708fdbc6c078db05b6104105b36cf2dbc6e85480d3cecab50e4e2750c0402a0f7f8000300000f01025b2502085b25dead5b25beef030435623235051000112233445566778899aabbccddeeff --discriminator 3840 --passcode 20202021  --storage-path admin_storage.json   --tests test_TC_ACE_2_3 --timeout 10000

Log:

[TC-ACE-2.3_Contact Sensor.txt](https://github.com/user-attachments/files/20225781/TC-ACE-2.3_Contact.Sensor.txt)

### Bug prevalence

3 times in an hour

### GitHub hash of the SDK that was being used

Version: v2.13-beta3.3+summer2025 Sha: a12250a4e4ffce5be5f809479fbe8d85e5acd230  

### Platform

raspi

KishokG, 2025-05-16:
@ashley-se Could you please run the script with this arg: `--bool-arg ci_only_linux_skip_ota_cluster_disallowed_for_certification:True`

cecille, 2025-05-16:
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/AccessControlEnforcement.adoc#tc-ace-2-3-command-privilege-enforcement-dut_server

bzbarsky-apple, 2025-05-16:
That's UNSUPPORTED_COMMAND instead of UNSUPPORTED_ACCESS.   This sounds like it might be https://github.com/project-chip/connectedhomeip/issues/37342?

tcarmelveilleux, 2025-05-20:
> That's UNSUPPORTED_COMMAND instead of UNSUPPORTED_ACCESS. This sounds like it might be [#37342](https://github.com/project-chip/connectedhomeip/issues/37342)?

Yes correct, but at the same time the command should be there if the cluster is there.

tcarmelveilleux, 2025-05-20:
The issue is that the OTA Software Update Requestor cluster is there but not hooked-up. This is not a bug with the test. The problems listed at the end give the information needed.

closed 2025-05-20 by tcarmelveilleux