project-chip/connectedhomeip #39001: AccessingFabricIndex should be available for all commands

gmarcosb, 2025-05-15:
### Reproduction steps

Currently it seems that there are conditions (`!mGoneAsync`) for [the context fabric ID to be available while handling a command](https://github.com/project-chip/connectedhomeip/blob/9b54b6708dee9d4d74839173ec824d78f460fd6c/src/app/CommandHandlerImpl.cpp#L762)

This causes problems with fabric-scoped structures, where we need to always know the fabric ID so that it can be propagated to the fabric-scoped structure.

What is the context here? Shouldn't there always be a way to find out what the fabric id is for a command being handled?

### Bug prevalence

Unsure

### GitHub hash of the SDK that was being used

head

### Platform

core

bzbarsky-apple, 2025-05-16:
@gmarcosb Once you go async, a fabric index is meaningless: by the time you are doing stuff that same fabric index could represent a different fabric, no?

gmarcosb, 2025-05-20:
Concerns addressed @ https://github.com/project-chip/connectedhomeip/pull/39070 - feel free to close when that goes in

closed 2025-05-20 by mergify[bot]