project-chip/connectedhomeip #39052: [SQA] Ble-wifi commissioning gets fail while trying Lock-app as DUT in RPI platform(Linux)

KishokG, 2025-05-19:
### Reproduction steps

Built lock-app and chiptool in two seperate RPI or install the TH image Version: v2.13-beta3.3+summer2025, use the sample app available in the apps folder.

**`Note`**: to build the TH image, Follow the below instruction:

> **Fresh installation:**
> git clone -b v2.13-beta3.3+summer2025 https://github.com/project-chip/certification-tool.git --recurse-submodules
> cd ~/certification-tool
> ./scripts/pi-setup/auto-install.sh
> At the end of the script, select option 1 to restart the RaspberryPi.
> Wait about 10 minutes.

> Update from previous versions:
> cd ~/certification-tool
> ./scripts/ubuntu/auto-update.sh v2.13-beta3.3+summer2025
> ./scripts/start.sh

- Brought up DUT using commnad : ./chip-lock-app --wifi --discriminator <3822>
- Tried pairing CNTRL using ./chip-tool pairing ble-wifi 1 SSID SSID-PWD 20202021 3822

**Expected behavior:** Lock-app and Chip-tool should be commission over Ble-wifi transport without getting any error.

**Actual behavior:**

> [1747638512.451] [106760:106766] [DMG] ReportDataMessage =
> [1747638512.451] [106760:106766] [DMG] {
> [1747638512.451] [106760:106766] [DMG]  AttributeReportIBs =
> [1747638512.451] [106760:106766] [DMG]  [
> [1747638512.451] [106760:106766] [DMG]          AttributeReportIB =
> [1747638512.451] [106760:106766] [DMG]          {
> [1747638512.452] [106760:106766] [DMG]                  AttributeDataIB =
> [1747638512.452] [106760:106766] [DMG]                  {
> [1747638512.452] [106760:106766] [DMG]                          DataVersion = 0xe298a902,
> [1747638512.452] [106760:106766] [DMG]                          AttributePathIB =
> [1747638512.452] [106760:106766] [DMG]                          {
> [1747638512.452] [106760:106766] [DMG]                                  Endpoint = 0x0,
> [1747638512.452] [106760:106766] [DMG]                                  Cluster = 0x46,
> [1747638512.452] [106760:106766] [DMG]                                  Attribute = 0x0000_0002,
> [1747638512.452] [106760:106766] [DMG]                          }
> [1747638512.452] [106760:106766] [DMG]
> [1747638512.452] [106760:106766] [DMG]                          Data = 5000 (unsigned),
> [1747638512.452] [106760:106766] [DMG]                  },
> [1747638512.452] [106760:106766] [DMG]
> [1747638512.453] [106760:106766] [DMG]          },
> [1747638512.453] [106760:106766] [DMG]
> [1747638512.453] [106760:106766] [DMG]          AttributeReportIB =
> [1747638512.453] [106760:106766] [DMG]          {
> [1747638512.453] [106760:106766] [DMG]                  AttributeDataIB =
> [1747638512.453] [106760:106766] [DMG]                  {
> [1747638512.453] [106760:106766] [DMG]                          DataVersion = 0xe298a902,
> [1747638512.453] [106760:106766] [DMG]                          AttributePathIB =
> [1747638512.453] [106760:106766] [DMG]                          {
> [1747638512.453] [106760:106766] [DMG]                                  Endpoint = 0x0,
> [1747638512.453] [106760:106766] [DMG]                                  Cluster = 0x46,
> [1747638512.453] [106760:106766] [DMG]                                  Attribute = 0x0000_0001,
> [1747638512.453] [106760:106766] [DMG]                          }
> [1747638512.454] [106760:106766] [DMG]
> [1747638512.454] [106760:106766] [DMG]                          Data = 300 (unsigned),
> [1747638512.454] [106760:106766] [DMG]                  },
> [1747638512.454] [106760:106766] [DMG]
> [1747638512.454] [106760:106766] [DMG]          },
> [1747638512.454] [106760:106766] [DMG]
> [1747638512.454] [106760:106766] [DMG]          AttributeReportIB =
> [1747638512.454] [106760:106766] [DMG]          {
> [1747638512.454] [106760:106766] [DMG]                  AttributeDataIB =
> [1747638512.454] [106760:106766] [DMG]                  {
> [1747638512.454] [106760:106766] [DMG]                          DataVersion = 0xe298a902,
> [1747638512.455] [106760:106766] [DMG]                          AttributePathIB =
> [1747638512.455] [106760:106766] [DMG]                          {
[... 146 lines cut ...]

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

a12250a4e4ffce5be5f809479fbe8d85e5acd230

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

### Anything else?

The All-clusters-app is pairing successfully with Chip-tool, at the same time Lock-app is not commissioning with chip-tool in Ble-wifi transport.
So it seems, there is a bug in Lock-app.

**DUT Log(Ble-wifi):** [Wifi_commissioning_failure_lock app(DUT).txt](https://github.com/user-attachments/files/20279602/Wifi_commissioning_failure_lock.app.DUT.txt)

**Controller Log(ble-wifi):** [Wifi_commissioning_failure_lock app(chip-tool).txt](https://github.com/user-attachments/files/20279611/Wifi_commissioning_failure_lock.app.chip-tool.txt)

bzbarsky-apple, 2025-05-20:
Probably this part:
```
 [1747638512.462] [106760:106766] [CTL] Failed to read Wi-Fi ConnectMaxTimeSeconds (endpoint 0): src/app/ClusterStateCache.cpp:379: CHIP Error 0x00000010: Key not found
```

bzbarsky-apple, 2025-05-30:
Also, see https://github.com/project-chip/connectedhomeip/issues/30184