project-chip/connectedhomeip #39128: phase Out %.*s in logs for compatibility with updated pw_tokenizer

Alami-Amine, 2025-05-22:
- Rebasing to newer pigweed versions will not be possible after the commit: https://github.com/google/pigweed/commit/285091bd42df9f15162fee653989dee51553e557

- basically `pw_tokenizer` which we use in our logging stopped allowing ` %.*s` 

- We basically need to phase out All usage of  ` %.*s` in strings such as below:
```cpp
ChipLogError(Zcl, "Invalid country code: '%.*s'", static_cast<int>(location.size()), location.data());
```

drippypop, 2025-05-22:
Hello, can I work on this?

bzbarsky-apple, 2025-05-22:
@Alami-Amine What does pigweed propose as a replacement for logging non-null-terminated strings?

bzbarsky-apple, 2025-05-22:
Looking at the pigweed issue, the problem is that right now they actually handle `%.*` incorrectly, so with that logging backend (but not others), those logs are broken anyway?

255, 2025-06-11:
Pigweed's recommended workaround is to use `pw::InlineString` to create a local null-terminated copy of the string. For example:

```c++
ChipLogError(Zcl, "Invalid country code: '%s'", pw::InlineString<5>(location).c_str());
```

See pw_tokenizer's [length-delimited strings](https://pigweed.dev/pw_tokenizer/tokenization.html#length-delimited-strings) docs for details.

`std::format` would work, but would requirre dynamic allocations and incur a significant code size cost.

255, 2025-06-12:
The non-type template parameter for `pw::InlineString` is the maximum size of the string, not the actual size. You can pick any value that you know will be large enough, e.g.
```c++
ChipLogError(Zcl, "Invalid country code: '%s'", pw::InlineString<100>(location).c_str());
```
or
```c++
ChipLogError(Zcl, "Invalid country code: '%s'", pw::InlineString<kMaxCountryCodeLength>(location).c_str());
```

The situation is the same as with `chip::StringBuilder<100>` in your example, where `100` is an arbitrary "large enough" value. Personally, I would recommend `pw::InlineString` over `chip::StringBuilder` for this situation since it will be much lower overhead.

bzbarsky-apple, 2025-06-14:
You may or may not be able to do pw::InlineString, because not all our logging backends are pigweed, and not everything links to pigweed....

arielsz71, 2025-06-17:
Yes, I wrote in the Maintainability chat about that last Friday, and using the StringBuilder approach seems more appropriate for now, because pigweed libraries do not always link easily and will require many changes in BUILD.gn files, so using the StringBuilder AddFormat() approach with some wrapper would be the way to go.

arielsz71, 2025-06-24:
Just to let you know that I am making good progress in my draft PR to solve this issue, I am waiting to see if the build passes on all targets before making the final adjustments, this is my first assignment and I am getting familiar with the code base, I tried many different ways to solve this until finding a way more in sync with the current code base, that is why it is taking a little more time than expected.

Alami-Amine, 2025-07-22:
Thank you @arielsz71!

- [ ] I see that you still have a post-merge comment, can you please address it before we close this ticket? https://github.com/project-chip/connectedhomeip/pull/39629#discussion_r2213320165