project-chip/connectedhomeip #39173: [BUG] Missing Null Pointer Check in ReadRevisionAttribute Function

agatah2333, 2025-05-25:
### Reproduction steps

A potential crash vulnerability has been identified in the `ReadRevisionAttribute` function within the Channel, Account Login, TargetNavigator, ContentLauncher, and MediaPlayback attribute access paths. The delegate parameter lacks proper null pointer validation. The following content uses the Channel cluster as a concrete example.

## Steps:

### 1. **Build the All Clusters App**
Since the current All Clusters App does not support the related cluster, we followed the instructions in [this issue](https://github.com/project-chip/connectedhomeip/issues/37889).

### 2. **Use chip-tool to read the revision attribute in the related cluster**
```bash
./chip-tool any read-by-id 0x0504 0x0000FFFD 0x654324 1

The All Clusters App shows the following log (as attached):
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==50919==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x5605cf8f9361 bp 0x7fffac62e750 sp 0x7fffac62e680 T0)
```

After checking the code, **a comparison with other attribute access paths** reveals an inconsistency:

- ChannelList Attribute Access: Includes isDelegateNull(delegate, endpoint) check

- LineupInfo Attribute Access: Includes isDelegateNull(delegate, endpoint) check

- ReadRevisionAttribute Function: Missing null pointer check

Most attribute/command access paths include null delegate validation, but the ReadRevisionAttribute() path has omitted this critical validation step.

## Reproduction Status

✅ Successfully Reproduced: Issue confirmed in our self-built All Clusters App

[app-error.txt](https://github.com/user-attachments/files/20432875/app-error.txt)

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

ab3d5ae

### Platform

core

### Platform Version(s)

v1.4

bzbarsky-apple, 2025-05-27:
Ideally, the revision would come from the metadata, not the delegate....

closed 2026-08-19 by mergify[bot]

Alami-Amine, 2026-09-16:
- This is reachable only when an application enables these clusters (Channel, AccountLogin, TargetNavigator, ContentLauncher, MediaPlayback) in ZAP without registering a delegate for them via `SetDefaultDelegate`. 
  - A conformant product implementing any of these clusters must supply a delegate — that is where the cluster's actual behavior lives, so a product without one would have no functioning implementation to ship.
  - Therefore this does not affect a correctly integrated device.

- Even in that case, the path to `ReadRevisionAttribute` sits behind the Interaction Model's access-control check. 
  - Interaction Model exchanges require an encrypted session (an unsecured ReadRequest is rejected before an exchange is created), and an unauthenticated session carries no subject identity, so it matches no ACL entry.
  -  The read therefore requires an authenticated peer. The CVE's "remote unauthenticated attacker" characterization is not accurate.