project-chip/connectedhomeip #39174: [Test Failed For TC_ACE_2.1] UnsupportedAccess for PreferredExtendedPanID and ThreadNetworks attributes at network-manager-app example

JasonLan0223, 2025-05-26:
### Test issue(s)

1. DUT running Linux example for matter-network-manager-app
2. Raspberry Pi running certification-tool at branch v2.11+fall2024
3. Running python Test case TC_ACE_2.1 and it will show fail for PreferredExtendedPanID and ThreadNetworks attributes
...

### Platform

raspi

### Anything else?

At DUT side show AccessControl: denied for reading Cluster=0x0000_0453 AttributeId=0x0000_0000 and AttributeId=0x0000_0001.
[1746523968.893360][7450:7450] CHIP:DMG: Reading attribute: Cluster=0x0000_0453 Endpoint=1 AttributeId=0x0000_0000 (expanded=1)
[1746523968.893374][7450:7450] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t=0x00010001 c=0x0000_0453 e=1 p=m r=r
[1746523968.893396][7450:7450] CHIP:DMG: AccessControl: allowed
        
[1746523968.893586][7450:7450] CHIP:DMG: Reading attribute: Cluster=0x0000_0453 Endpoint=1 AttributeId=0x0000_0001 (expanded=1)
[1746523968.893600][7450:7450] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t=0x00010001 c=0x0000_0453 e=1 p=o r=r
[1746523968.893622][7450:7450] CHIP:DMG: AccessControl: allowed

JasonLan0223, 2025-06-02:
@bzbarsky-apple ,
yes. it is failed for Linux network-manager-app. It seems the privilege is not correct for two PreferredExtendedPanID and ThreadNetworks attributes. Could you have any suggestion?

JasonLan0223, 2025-06-02:
@bzbarsky-apple , 
We found the PreferredExtendedPanID and ThreadNetworks of data-model access is not view access for thread-network-directory-cluster.xml. After we change it from manage to view, then it could PASS for TC_ACE_2.1 test case. Could you review is it correct to fix this issue for our attach? Thanks. 

[001-TC_ACE_2_1-FailForAccessControl.patch](https://github.com/user-attachments/files/20544582/001-TC_ACE_2_1-FailForAccessControl.patch)

bzbarsky-apple, 2025-06-05:
@ksperling-apple Looks like the XML is not matching the spec adoc here.  Which one is right?

closed 2025-06-09 by mergify[bot]