project-chip/connectedhomeip #39182: [1.3] TH v2.10.2+spring2024] Door Lock Tests Fail Due to Fixed PIN Code and CredentialData Length in Test Harness

bilalahmaddev, 2025-05-26:
### Reproduction steps

I'm using the Matter 1.3 certification tool branch with Test Harness version v2.10.2+spring2024, and I encountered an issue with the Door Lock cluster tests.

The test scripts in this harness version use a hardcoded PIN code length of 6, which does not respect the MinPINCodeLength and MaxPINCodeLength attributes reported by the device. As a result, two tests were consistently failing.

After significant debugging and reviewing logs, I updated my TC_DRLK_2.1.yaml test scripts to fetch and use the actual MinPINCodeLength and MaxPINCodeLength attributes dynamically. This resolved the issue, and the tests passed successfully.

I noticed that this issue has already been addressed and merged in the Connected Home IP repo:
[CSA Commit – Fix Door Lock Test Code Length](https://github.com/project-chip/connectedhomeip/pull/35182/commits/7ff26a68d919c3a2641ce298685c1418f1a1632f)

However, this fix does not appear to be included in the current certification test harness (v2.10.2+spring2024).

How to make sure if ATL uses the updated test scripts? 

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

v2.10.2+spring2024,

### Platform

raspi

### Platform Version(s)

1.3

### Type

Platform Issue

bzbarsky-apple, 2025-05-27:
@cjandhyala

KishokG, 2025-05-30:
Hi @bilalahmaddev In the commit(https://github.com/project-chip/connectedhomeip/pull/35182/commits/7ff26a68d919c3a2641ce298685c1418f1a1632f) which you shared above is just removing the precondition and adding the same steps for set user(Step 1d), set credential(Step 1g), reading MinPINCodeLength(Step 1e) and MaxPINCodeLength(Step 1f). And the value of MinPINCodeLength and MaxPINCodeLength is not used in any other steps. Can you please share more details about it? Like what did you modified in the script? which test step? Also please share the failure logs.

bilalahmaddev, 2025-05-30:
Our device use max 4 length pin code and test TC_DRLK_2.1 use 6 bytes fixed length pin code and credential. that is why test is failing, test is using 6 bytes fixed length code, it should be used according MaxPINCodeLength attribute of DUT.