project-chip/connectedhomeip #39212: [BUG] Failed to perform Matter device setup setup: Error Domain=HMErrorDomain Code=2

LucaFagan, 2025-05-27:
### Reproduction steps

We have the following configuration:
- our thread border router
- our matter device
- our app used to commission the device into our fabric

We commission matter device in our fabric, using system Matter.framework and MatterSupport.framework.

We have created Matter extension that subclasses MatterAddDeviceExtensionRequestHandler.

In Matter Extension the following callbacks are fired:

override func selectThreadNetwork(from threadScanResults: [MatterAddDeviceExtensionRequestHandler.ThreadScanResult]) async throws -> MatterAddDeviceExtensionRequestHandler.ThreadNetworkAssociation

override func commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) async throws

**_Everything worked until a few weeks ago._**

Now when we call request.perform() we always get the error: 

> [0CA8C59B] Failed to perform Matter device setup setup: Error Domain=HMErrorDomain Code=2 "(null)" UserInfo={HFErrorUserInfoOptionsKey={
>     HFErrorUserInfoOptionDescriptionKey = "L\U2019app Casa non \U00e8 riuscita a connettersi all\U2019accessorio.";
>     HFErrorUserInfoOptionTitleKey = "Impossibile aggiungere l\U2019accessorio";
> }} 

[image]

The commissioning is done in this way:

`        let homes = [MatterAddDeviceRequest.Home(displayName: "MyHome")]
        let topology = MatterAddDeviceRequest.Topology(ecosystemName: Consts.App.ecosystemName, homes: homes)

if let setupPayload = try? MTRSetupPayload(onboardingPayload: self.container.payload!)
{
            var request = MatterAddDeviceRequest(topology: topology)
            request.setupPayload = setupPayload

            let callbackQueue = DispatchQueue(label: "com.bhtronics.pairing.callback")
            self.matterController.setDeviceControllerDelegate(self, queue: callbackQueue)

              do
              {
				try await request.perform()
			  }

              catch let error
              {
              	print(error.localizedDescription)
             }
}`

The reported error seems to be related to Apple Home app.

I've tried everything to resolve:
- restart iOS device
- restart router, border router and matter device
- removing all keys in MatterStorage
- reset network connections
- uninstall and reinstall our app

But we keep getting the error in some iOS devices (some with ios 18.4.1 and other with 18.5). The weird thing is that until a few weeks ago everything was working.

How can I check what's wrong? 
Has anything been changed on the Apple platform?

I've attached ConsoleLog.txt.

Regards
Luca Fagan

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

system Matter.framework and MatterSupport.framework

### Platform

darwin

### Anything else?

[ConsoleLog.txt](https://github.com/user-attachments/files/20457218/ConsoleLog.txt)

bzbarsky-apple, 2025-05-27:
@LucaFagan This does not sound like an issue with the open-source Matter SDK.  Please report the problem to Apple following the steps at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple

closed 2025-05-27 by bzbarsky-apple

LucaFagan, 2025-06-09:
@bzbarsky-apple I've reported the problem to Apple on May 30, (feedback id is FB17743117) but so far no one has answered me. Can you please check if someone has picked up my ticket?