project-chip/connectedhomeip #39249: BDX Log Retrieval Fails with kBusy Error After Inline Log Transfer in Diagnostic Logs Cluster

Saravana-kr22, 2025-05-28:
**Description**:
We are encountering an issue while testing the Diagnostic Logs Cluster where fetching BDX logs using the python controller is fails with a kBusy error after successfully reading inline logs.
 
**Observed Behavior**:

- When I retrieve only inline logs, the logs are fetched successfully.

- When I retrieve only BDX logs, the transfer is also successful.

- However, when I first retrieve inline logs, followed by an attempt to retrieve BDX logs in the same session:

- The BDX transfer fails with a kBusy error.

- Only the inline log retrieval continues to work.

- The test case ends with a core dump.

 
GitHub hash of the SDK that were being used are : https://github.com/project-chip/connectedhomeip/commit/f8f3c4977d756c81f5dd61bd6eea10448098ba2c

**Expected Behavior**:

Sequential or back-to-back retrieval of logs via both inline and BDX mechanisms should be supported. The BDX transfer should not fail with a kBusy error following a successful inline transfer

Log: [bdx_core_dump_error.txt](https://github.com/user-attachments/files/20489396/bdx_core_dump_error.txt)

bzbarsky-apple, 2025-05-28:
@Saravana-kr22 It's very hard to tell what _exact_ commands are being sent from client to server here, or what server is being used.  Can you reproduce the issue with chip-tool sending RetriveLogsRequest commands to the server?  If so, can you list the exact chip-tool commands that are being used?

If this is not reproducible with chip-tool, please provide explicit, detailed steps to reproduce this issue.  Assume that the SDK repository has been cloned and SHA https://github.com/project-chip/connectedhomeip/commit/f8f3c4977d756c81f5dd61bd6eea10448098ba2c checked out.  What does on do after that to reproduce this?

bzbarsky-apple, 2025-05-31:
OK, so I tried doing the following:

```
dd if=/dev/zero of=/tmp/end_user_support_log.txt bs=256 count=2
dd if=/dev/zero of=/tmp/crash_log.txt bs=1024 count=11  
dd if=/dev/zero of=/tmp/network_log.txt bs=1024 count=11 

scripts/examples/gn_build_example.sh examples/all-clusters-app/linux out/debug/standalone is_asan=true

./out/debug/standalone/chip-all-clusters-app --end_user_support_log /tmp/end_user_support_log.txt --network_diagnostics_log /tmp/network_log.txt --crash_log /tmp/crash_log.txt
```

and in a separate terminal, `chip-tool interactive start` followed by:

```
pairing code 1 MT:-24J042C00KA0648G00

diagnosticlogs retrieve-logs-request 0 1 1 0 --TransferFileDesignator Length_123456789123456789123.txt

diagnosticlogs retrieve-logs-request 2 1 1 0 --TransferFileDesignator Length_123456789123456789123.txt
```

That first command responds with a status of 1 (Exhausted), as expected.

The second command works fine and does a BDX transfer, saving things to `/tmp/Length_123456789123456789123.txt`.

I don't get any Busy responses @suhailab.

Note that if you try to do another transfer _with the same file designator_, the client will error out with:

```
[1748680006.829] [98038:41256352:chip] [TOO] The file '/tmp/Length_123456789123456789123.txt' for dumping the logs already exists.
[1748680006.829] [98038:41256352:chip] [-] examples/chip-tool/commands/common/BDXDiagnosticLogsServerDelegate.cpp:88: CHIP Error 0x00000003: Incorrect state at src/protocols/bdx/BdxTransferDiagnosticLog.cpp:175
```

but that's just the chip-tool behavior: it wants the file to not exist to do the transfer.

bzbarsky-apple, 2025-05-31:
Ah, OK, so that matches what you see @suhailab : chip-tool does not show the problem.

So back to: what are the _exact_ steps to reproduce?  Starting with `git clone` of the Matter repository.  Because it's a waste of time for me to try to guess what you are running....

But it sure sounds like the Python bits are broken somewhere and should be fixed.

suhailab, 2025-06-03:
Hi @bzbarsky-apple 

Below is the standalone code to reproduce the bdx_file transfer fail issue . I am running it using the following steps:
    1) We need to open two terminals T1 and T2, one for DUT(T1)--for running chip-all-cluster-app and  other one (T2) for 
        controller--Running the test case.
    
    2) In T2, activate the Python environment we would use for running the python cert tests like TC_DRLK_2_13.py.
    
    3) In T1 terminal, use below command to advertise the chip-all-cluster-app:
         ```
rm -rf /tmp/chip_* && ./chip-all-clusters-app --trace_decode 1 \
        --end_user_support_log end_user.txt \
        --network_diagnostics_log network_log.txt \
        --crash_log crash_log.txt \
        --discriminator 3456
```
    
    4) In T2 terminal, run the below python test case to fetch the different diagnostic type logs:
    ` python test_diag_bdx_logs.py --discriminator 3456 --passcode 20202021 --commissioning-method on-network`

The standalone test script (test_diag_bdx_logs.py) is attached below to help reproduce the issue directly.

[test_diag_bdx_logs.zip](https://github.com/user-attachments/files/20575870/test_diag_bdx_logs.zip)

cecille, 2025-06-05:
@harimau-qirex - would you be able to take a quick look here?

closed 2025-08-08 by mergify[bot]