project-chip/connectedhomeip #39390: Generate `event info` metadata for use in clusters

andy31415, 2025-06-04:
Make it easy to create "correct implementations" for event info metadata, specifically having non-view accesss rights based on eventID.

We likely need to codegen some form of metadata for events per cluster and then correctly pick up the readPrivileges for them.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/39370#discussion_r2127082756_

ratgr, 2025-07-18:
@andy31415 Now the good question is, how should we use this Info in the default provider, using it from the Default provider will need to generate a replacement for provider->EventInfo that uses this info, or maybe just remove the default, and each cluster provide its own so as not to include all un-necesary information?

These are some of the options I see: 

### template DefaultServerCluster by the Cluster
Add a template parameter with the ClusterInfo we need for solving the metadata to add to the info
Pros:
-Similar to CHI implementation
- No extra pointer indirection
Cons:
Will generate a unique class per cluster used

### DefaultServerCluster MetadataResolution pointer
At construction time expect a "Resolution" function pointer to solve the metadata which would point to a templated function on the cluster
Pros:
- Keeps DefaultServerCluster Monomorphic no template
Cons:
- You still need to provide a template parameter either in an argument constructor or in the constructor itself

### Provide all event encompasing solution since the begining
Pros: 
- Simplest
- No templates
Cons:
- If we end up having all the events then flash might get used

I think the option of using an extra indirection for resolution is the best option and it should make the .text sector the smallest

ratgr, 2025-07-18:
rewriting a use case in the 2 versions that need to be updated
```
class OtaProviderServer : public DefaultServerCluster, private OtaProviderLogic
{
public:
    OtaProviderServer(EndpointId endpointId) : DefaultServerCluster({ endpointId, OtaSoftwareUpdateProvider::Id }) {}
   ...
};
```

Templated Default cluster
```
class OtaProviderServer : public DefaultServerCluster<OtaSoftwareUpdateProvider::Id >, private OtaProviderLogic
{
public:
    OtaProviderServer(EndpointId endpointId) : DefaultServerCluster({ endpointId, OtaSoftwareUpdateProvider::Id }) {}
   ...
};
```

MetadataResolution pointer
```
class OtaProviderServer : public DefaultServerCluster<OtaSoftwareUpdateProvider::Id >, private OtaProviderLogic
{
public:
    OtaProviderServer(EndpointId endpointId) : DefaultServerCluster({ endpointId, OtaSoftwareUpdateProvider::Id }, MetadataProvider<EventEntry, OtaSoftwareUpdateProvider::Id> ) {}
   ...
};
```