project-chip/connectedhomeip #39494: [darwin] flaky test, leaks in test001_BasicControllerStartup (and this seems to make other tests fail)

andy31415, 2025-06-11:
https://github.com/project-chip/connectedhomeip/actions/runs/15590472509/job/43908928430 says:

```
2025-06-11T17:09:02.2390620Z leaks Report Version: 4.0
2025-06-11T17:09:02.2423810Z Process 29600: 91708 nodes malloced for 6993 KB
2025-06-11T17:09:02.2596530Z Process 29600: 28 leaks for 8096 total leaked bytes.
2025-06-11T17:09:02.2597210Z 
2025-06-11T17:09:02.2861100Z     28 (7.91K) << TOTAL >>
2025-06-11T17:09:02.2861640Z 
2025-06-11T17:09:02.2862550Z       7 (4.31K) ROOT LEAK: <NSArray 0x600003e52950> [16]
2025-06-11T17:09:02.2864370Z          6 (4.30K) __strong _object --> <XCTestExpectation 0x600001759b90> [112]
2025-06-11T17:09:02.2865970Z             2 (2.05K) __strong _creationCallStackReturnAddresses --> <_NSCallStackArray 0x6000032e9980> [48]
2025-06-11T17:09:02.2867550Z                1 (2.00K) _frames --> <_NSCallStackArray._frames (malloc[]) 0x7f8da5836600> [2048]
2025-06-11T17:09:02.2869530Z             2 (2.05K) __strong _fulfillCallStackReturnAddresses --> <_NSCallStackArray 0x600002ac0090> [48]
2025-06-11T17:09:02.2871060Z                1 (2.00K) _frames --> <_NSCallStackArray._frames (malloc[]) 0x7f8da481d400> [2048]
2025-06-11T17:09:02.2872380Z             1 (96 bytes) __strong _cleanupHandlersLock --> <NSLock 0x600001859320> [96]
2025-06-11T17:09:02.2873840Z 
2025-06-11T17:09:02.2874270Z       7 (2.27K) ROOT LEAK: <_XCTestCaseInterruptionException 0x600002ac0660> [48]
2025-06-11T17:09:02.2875620Z          5 (2.17K) reserved --> <NSMutableDictionary 0x60000373c4e0> [32]  item count: 2
2025-06-11T17:09:02.2876790Z             4 (2.14K) <NSMutableDictionary (Storage) 0x600002ac05a0> [48]
2025-06-11T17:09:02.2877750Z                2 (2.05K) <_NSCallStackArray 0x600002ac05d0> [48]
2025-06-11T17:09:02.2881960Z                   1 (2.00K) _frames --> <_NSCallStackArray._frames (malloc[]) 0x7f8da481dc00> [2048]
2025-06-11T17:09:02.2883590Z                1 (48 bytes) <_NSCallStackArray 0x600002ac0600> [48]
2025-06-11T17:09:02.2885820Z          1 (48 bytes) reason --> <CFString 0x600002ac0630> [48]  length: 17  "Interrupting test"
2025-06-11T17:09:02.2895960Z 
2025-06-11T17:09:02.2896330Z       9 (880 bytes) ROOT LEAK: <MTRDevice_Concrete 0x7f8da4404080> [496]
2025-06-11T17:09:02.2897610Z          1 (128 bytes) __strong _queue --> <OS_dispatch_queue_serial 0x600001272980> [128]  "org.csa-iot.matter.framework.device.workqueue" (from Matter)
2025-06-11T17:09:02.2898990Z          2 (96 bytes) __strong _asyncWorkQueue --> <MTRAsyncWorkQueue 0x6000032eb990> [48]
2025-06-11T17:09:02.2899960Z             1 (48 bytes) __strong _items --> <NSMutableArray 0x6000032eb570> [48]  item count: 0
2025-06-11T17:09:02.2900930Z          2 (64 bytes) __strong _delegates --> <NSMutableSet 0x600003deb4c0> [32]  item count: 0
2025-06-11T17:09:02.2903960Z             1 (32 bytes) <NSMutableSet (Storage) 0x600003d90be0> [32]
2025-06-11T17:09:02.2905770Z          1 (32 bytes) __strong _expectedValueCache --> <NSMutableDictionary 0x600003de82e0> [32]  item count: 0
2025-06-11T17:09:02.2907640Z          1 (32 bytes) __strong _matterCPPObjectsHolder --> <MTRDeviceMatterCPPObjectsHolder 0x600003dea140> [32]
2025-06-11T17:09:02.2909480Z          1 (32 bytes) __strong _persistedClusters --> <NSMutableSet 0x600003dea420> [32]  item count: 0
2025-06-11T17:09:02.2910320Z 
2025-06-11T17:09:02.2910840Z       2 (192 bytes) ROOT LEAK: <MTRDeviceTestDelegate 0x600000e406c0> [144]
2025-06-11T17:09:02.2912930Z          1 (48 bytes) __strong _onReachable --> <__NSMallocBlock__ 0x600003223db0> [48]  MatterTests  __45-[MTRDeviceTests test049_CorrectTimeOnDevice]_block_invoke  0x109e7b180  /Users..."
2025-06-11T17:09:02.2914630Z 
2025-06-11T17:09:02.2914970Z       1 (160 bytes) ROOT LEAK: <CFRunLoopObserver 0x600000d5c1e0> [160]
2025-06-11T17:09:02.2916000Z       1 (96 bytes) ROOT LEAK: <MTRSetupPayload 0x600001859740> [96]
2025-06-11T17:09:02.2916990Z       1 (32 bytes) ROOT LEAK: <MTRTestControllerDelegate 0x600003debe40> [32]
2025-06-11T17:09:02.2917700Z 
2025-06-11T17:09:02.4035410Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestCase.mm:162: error: -[MTRPerControllerStorageTests test001_BasicControllerStartup] : failed - LEAKS DETECTED
2025-06-11T17:09:02.4115390Z Test Case '-[MTRPerControllerStorageTests test001_BasicControllerStartup]' failed (17.876 seconds).

...

2025-06-11T17:18:21.0967450Z Test session results, code coverage, and logs:
2025-06-11T17:18:21.0968500Z 	/tmp/darwin/framework-tests/TestResults.xcresult
2025-06-11T17:18:21.0969300Z 
2025-06-11T17:18:21.0981930Z 	-[MTRDeviceTests test049_CorrectTimeOnDevice]
2025-06-11T17:18:21.0983310Z 	-[MTRPerControllerStorageTests test001_BasicControllerStartup]
2025-06-11T17:18:21.0984670Z 	-[MTRPerControllerStorageTests test002_TryStartingTwoControllersWithSameNodeID]
2025-06-11T17:18:21.0986080Z 	-[MTRPerControllerStorageTests test003_TestTwoControllersSameUUID]
2025-06-11T17:18:21.0987300Z 	-[MTRPerControllerStorageTests test004_TestBasicSessionResumption]
2025-06-11T17:18:21.0989440Z 	-[MTRPerControllerStorageTests test005_TestSessionResumptionDataClearingNodeIDChanged]
2025-06-11T17:18:21.0990780Z 	-[MTRPerControllerStorageTests test006_TestSessionResumptionDataClearingCATsChanged]
2025-06-11T17:18:21.0991790Z 	-[MTRPerControllerStorageTests test007_TestMultipleControllers]
2025-06-11T17:18:21.0992560Z 	-[MTRPerControllerStorageTests test008_TestDataStoreDirect]
2025-06-11T17:18:21.0993330Z 	-[MTRPerControllerStorageTests test009_TestDataStoreMTRDevice]
2025-06-11T17:18:21.0994560Z 	-[MTRPerControllerStorageTests test010_TestDataStoreMTRDeviceWithBulkReadWrite]
2025-06-11T17:18:21.1005100Z 	-[MTRPerControllerStorageTests test011_TestDataStoreMTRDeviceWithStorageBehaviorOptimizationDisabled]
2025-06-11T17:18:21.1007390Z 	-[MTRPerControllerStorageTests test012_startSuspended]
2025-06-11T17:18:21.1009010Z 	-[MTRPerControllerStorageTests test013_suspendDevices]
2025-06-11T17:18:21.1010350Z 	-[MTRPerControllerStorageTests test014_TestDataStoreMTRDeviceInvalidateFlush]
2025-06-11T17:18:21.1011230Z 	-[MTRPerControllerStorageTests testControllerServer]
2025-06-11T17:18:21.1012630Z 	-[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingAttributes]
2025-06-11T17:18:21.1013680Z 	-[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingClusters]
2025-06-11T17:18:21.1014790Z 	-[MTRPerControllerStorageTests testDataStorageUpdatesWhenRemovingEndpoints]
2025-06-11T17:18:21.1015640Z 	-[MTRPerControllerStorageTests testMTRDeviceDealloc]
2025-06-11T17:18:21.1016610Z 	-[MTRPerControllerStorageTests testMTRDeviceMaybeUnreachableIgnoredIfReceivingFromDevice]
2025-06-11T17:18:21.1018030Z 	-[MTRPerControllerStorageTests testMTRDeviceResetSubscription]
2025-06-11T17:18:21.1019280Z 	-[MTRPerControllerStorageTests testSetMRPParametersWithRunningController]
2025-06-11T17:18:21.1020410Z 	-[MTRPerControllerStorageTests testSubscriptionPool]
2025-06-11T17:18:21.1021150Z 	-[MTRPerControllerStorageTests testSubscriptionPoolManyDevices]
2025-06-11T17:18:21.1021670Z 
2025-06-11T17:18:21.1021790Z ** TEST FAILED **
```

bzbarsky-apple, 2025-06-13:
The actual failure here is:

```
2025-06-11T17:08:05.7643730Z /Users/runner/work/connectedhomeip/connectedhomeip/src/darwin/Framework/CHIPTests/TestHelpers/MTRTestControllerDelegate.mm:57: error: -[MTRDeviceTests test049_CorrectTimeOnDevice] : ((error.code) equal to (0)) failed: ("16") is not equal to ("0")
```

The rest is knock-on effects; `test001_BasicControllerStartup` happens to be the first test that checks for leaks after that point, but the leaks are from the earlier test, as this line in the leak output shows:

```
2025-06-11T17:09:02.2912930Z          1 (48 bytes) __strong _onReachable --> <__NSMallocBlock__ 0x600003223db0> [48]  MatterTests  __45-[MTRDeviceTests test049_CorrectTimeOnDevice]_block_invoke  0x109e7b180  /Users..."
```

bzbarsky-apple, 2025-06-13:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/39477