project-chip/connectedhomeip #39498: [Tooling] Build failure on Apple Silicon Mac: "Bad CPU type in executable: 'zap-cli'" during Python build

Chapoly1305, 2025-06-11:
### Reproduction steps / Feature

On a new Mac mini, while rosetta 2 is not yet enabled, build process will fail. 
To replicate, 
```
scripts/checkout_submodules.py --platform darwin
source scripts/bootstrap.sh
source scripts/activate.sh
./scripts/build_python.sh
```

Here is the log.

```
alex@Alexs-Mac-mini connectedhomeip % scripts/checkout_submodules.py --platform darwin 
Checking out: nlassert, nlio, mbedtls, qrcode, pigweed, openthread, nanopb, third_party/jsoncpp/repo, editline, third_party/boringssl/repo/src, third_party/libwebsockets/repo, perfetto, third_party/abseil-cpp/src, third_party/fuzztest, third_party/googletest, third_party/re2/src
alex@Alexs-Mac-mini connectedhomeip % source scripts/bootstrap.sh 
2025-06-11 16:31:50,570 Loading extra packages for darwin
2025-06-11 16:31:50,570 Appending: /Users/alex/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json for this platform
2025-06-11 16:31:50,570 Skipping: windows (i.e. /Users/alex/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/python311.json)

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (1m0.1s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....done (1m15.9s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20250611 16:34:10 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Installing pip requirements for all...

[notice] A new release of pip is available: 23.2.1 -> 25.1.1
[notice] To update, run: pip install --upgrade pip
alex@Alexs-Mac-mini connectedhomeip % source scripts/activate.sh 
[... 90 lines cut ...]
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alex/connectedhomeip/.environment/cipd/packages/pigweed/lib/python3.11/subprocess.py", line 548, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alex/connectedhomeip/.environment/cipd/packages/pigweed/lib/python3.11/subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/alex/connectedhomeip/.environment/cipd/packages/pigweed/lib/python3.11/subprocess.py", line 1917, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 86] Bad CPU type in executable: 'zap-cli'
[977/1193] ACTION //:matter_build_venv._install_base_3p_deps(//third_party/pigweed/repo/pw_build/python_toolchain:python)
ninja: build stopped: subcommand failed.
```

### Platform

darwin

### Platform Version(s)

1.4

### Type

Manually tested with SDK

kiel-apple, 2025-06-13:
repro'd on macOS 15.5 arm64 with `connectedhomeip` commit `40b5fce`

Chapoly1305, 2026-02-14:
I think the issue was resolved in https://github.com/project-chip/zap/pull/1607, thereby I am closing this ticket.

closed 2026-02-14 by Chapoly1305