project-chip/connectedhomeip #39646: [Trivial] Consider taking the dptzStreams parameter by const& or & for consistency

yufengwangca, 2025-06-21:
### Type

Other

### Explanation

CHIP_ERROR LoadDPTZStreams(std::vector<DPTZStruct> dptzStreams) override               = 0;

Consider taking the dptzStreams parameter by const& or & for consistency with other Load... functions (e.g., LoadMPTZPresets). Passing by value can be inefficient due to the copy operation. Also, the current stub implementation in CameraAVSettingsUserLevelManager operates on a local copy, making dptzStreams.clear() ineffective for the caller.

### Platform

raspi

### Testing

CI tested

bzbarsky-apple, 2025-06-23:
This API is just broken, no?  The whole point is to mutate a caller-provided vector, and this API cannot do that.  I don't understnand why this is labeled "Trivial".

yufengwangca, 2025-09-05:
This has been addressed