project-chip/connectedhomeip #39667: ChipDeviceCtrl API documentation - Fix links

cecille, 2025-06-23:
The links in the TOC don't work in the compiled documentation:

https://project-chip.github.io/connectedhomeip-doc/testing/ChipDeviceCtrlAPI.html#

juandediosg, 2025-06-25:
### Issue
The generated Markdown documentation (`ChipDeviceCtrlAPI.md`) did not generate functional links in the table of contents (TOC) when generated a HTML via CMake build. This broke internal navigation.

### RCA (Root cause Analysis)
The original pydoc-markdown used the below config, that resulted in TOC links in the Markdown did not have proper ID anchors, causing broken navigation in the generated HTML.
```bash
insert_header_anchors: false
add_full_prefix: false
```

### Solution
Updated the pydoc-markdown configuration to automatically insert proper ID anchors, thus removing the awk command.
```bash
insert_header_anchors: true
add_full_prefix: false
```
Now the TOC entries show as `RegisterOnActiveCallback` instead of `chip.ChipDeviceCtrl.RegisterOnActiveCallback` with the anchor link like below
```bash
* [RegisterOnActiveCallback](#chip.ChipDeviceCtrl.RegisterOnActiveCallback)
```

#### Scope
Validate the generated documentation files are free of errors and broken references:

- **Markdown file**: Verify the generated `.md` file (e.g., `ChipDeviceCtrlAPI.md`) in `_build/src/testing`  and check
  - Includes valid ID anchor links. 
- **HTML output files**: Check the generated HTML files in `docs/_build/html/` and check:
  - Internal table of contents (TOC) links are working.
  - Anchor navigation jumps to the correct section.
  - No broken references.

juandediosg, 2025-06-25:
**PR Open:** Generate Documentation for python API: Fix TOC links
[#39695](https://github.com/project-chip/connectedhomeip/pull/39695) to fix the broken Table of Contents (TOC) links in the generated `ChipDeviceCtrlAPI.md` file.

juandediosg, 2025-06-26:
**Update summary:**

I have merged the latest changes from main branch into my branch.

Now, the PR **Generate Documentation for python API: Fix TOC links [#39695](https://github.com/project-chip/connectedhomeip/pull/39695)** is ready for Google code review.

Also, I have marked review comments as resolved and fixed the CI job issues, PR now shows:
- No merge conflicts
- All the CI checks passed

Moved the status to **Needs Google review**.

juandediosg, 2025-06-30:
Approved by Google, needs second company review.

juandediosg, 2025-07-08:
**Update Summary**

The PR **Generate Documentation for python API: Fix TOC links [#39695](https://github.com/project-chip/connectedhomeip/pull/39695)** includes the following update:

- Merged the latest changes from the `master` branch into `PyApi_Docs_Fix_links` to keep it up to date and resolve merge conflicts before integration.

closed 2025-07-11 by mergify[bot]

juandediosg, 2025-07-11:
**Update Summary**

The PR **Generate Documentation for python API: Fix TOC links [#39695](https://github.com/project-chip/connectedhomeip/pull/39695)** has been merged.