project-chip/connectedhomeip #39690: [BUG] local.py tests running against boringssl fail for LARGE_PAYLOAD

gmarcosb, 2025-06-24:
### Reproduction steps

Running:
```
./scripts/tests/local.py python-tests --test-filter TestCheckCommandFlags.py --keep-going
```

Get:

```
No memory at src/app/OperationalSessionSetup.cpp:254
```

The issue seems to be that [local.py uses `variants = ["no-ble", "clang", "boringssl"]`](https://github.com/project-chip/connectedhomeip/blob/31c8966f7903aacfeddb4d563a2336f30d16903e/scripts/tests/local.py#L76), and `boringssl` seems to be the underlying cause

### Bug prevalence

Workaround available by not using boringssl, but happens 100% with boringssl

### GitHub hash of the SDK that was being used

f3fc26aa7cecead92ddebac6230f8e0cca93515b

### Platform

core

bzbarsky-apple, 2025-06-24:
The obvious place that would come from is:

```
    mCASEClient = mClientPool->Allocate();
    VerifyOrReturnError(mCASEClient != nullptr, CHIP_ERROR_NO_MEMORY);
```

but that should not depend on boringssl....  This is very strange.

That said, when I run the command given in the steps to reproduce I get:

```
      scripts/run_in_python_env.sh: line 25: out/venv/bin/activate: No such file or directory
```

Presumably the full set of steps to reproduce is more extensive, and it would be good to know what those are.

gmarcosb, 2025-08-27:
Found the first issue; `CHIP_CONFIG_MAX_ACTIVE_TCP_CONNECTIONS` is set to 4 for `ChipDeviceCtrl`

closed 2025-09-09 by mergify[bot]