project-chip/connectedhomeip #39738: [CERT-TEST-FAILURE] [TC-SOIL-2.2] Python script failure

arulkumargrl, 2025-06-27:
### Feature Area

Other

### Test Case

TC-SOIL-2.2

### Reproduction steps

When we try to execute the TC-SOIL-2.2 Python script fails at test step 3,`TH reads from the DUT the SoilMoistureMeasuredValue attribute and save the value as measurement`  with below error. The failure occurs due to a mismatch in the expected value—the script expects a non-null value for SoilMoistureMeasuredValue, but the DUT returns it as null. 

> [MatterTest] 06-27 17:39:09.608 INFO 		***** Test Step 3 : Read SoilMoistureMeasuredValue attribute and save the value as measurement
> [MatterTest] 06-27 17:39:09.609 INFO Found an existing secure session to [1:0000000012344321]!
> [MatterTest] 06-27 17:39:09.610 INFO 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
> [MatterTest] 06-27 17:39:09.610 INFO <<< [E:3320i S:2370 M:47750045] (S) Msg TX from 000000000001B669 to 1:0000000012344321 [2341] [UDP:[fe80::d5cd:f44:3848:abfe%wlo1]:5540] --- Type 0001:02 (IM:ReadRequest) (B:52)
> [MatterTest] 06-27 17:39:09.611 INFO ??1 [E:3320i S:2370 M:47750045] (S) Msg Retransmission to 1:0000000012344321 scheduled for 352ms from now [State:Active II:500 AI:300 AT:4000]
> [MatterTest] 06-27 17:39:09.611 INFO >>> [E:3320i S:2370 M:66652541 (Ack:47750045)] (S) Msg RX from 1:0000000012344321 [2341] to 000000000001B669 --- Type 0001:05 (IM:ReportData) (B:70)
> [MatterTest] 06-27 17:39:09.615 INFO <<< [E:3320i S:2370 M:47750046 (Ack:66652541)] (S) Msg TX from 000000000001B669 to 1:0000000012344321 [2341] [UDP:[fe80::d5cd:f44:3848:abfe%wlo1]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
> [MatterTest] 06-27 17:39:09.615 ERROR Exception occurred in test_TC_SOIL_2_2.
> Traceback (most recent call last):
>   File "/home/grl/projectchip/masterconnectedhomeip/connectedhomeip/no/lib/python3.12/site-packages/mobly/base_test.py", line 801, in exec_one_test
>     test_method()
>   File "/home/grl/projectchip/masterconnectedhomeip/connectedhomeip/no/lib/python3.12/site-packages/chip/testing/decorators.py", line 366, in per_endpoint_runner
>     test_instance.event_loop.run_until_complete(asyncio.wait_for(
>   File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
>     return future.result()
>            ^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
>     return await fut
>            ^^^^^^^^^
>   File "/home/grl/projectchip/masterconnectedhomeip/connectedhomeip/src/python_testing/TC_SOIL_2_2.py", line 89, in test_TC_SOIL_2_2
>     asserts.assert_true(measurement != NullValue, "SoilMoistureMeasuredValue is NullValue")
>   File "/home/grl/projectchip/masterconnectedhomeip/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 319, in assert_true
>     fail(msg, extras)
>   File "/home/grl/projectchip/masterconnectedhomeip/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 441, in fail
>     raise signals.TestFailure(msg, extras)
> mobly.signals.TestFailure: Details=SoilMoistureMeasuredValue is NullValue, Extras=None
> [MatterTest] 06-27 17:39:09.619 INFO 		***** Test Failure : 
> [MatterTest] 06-27 17:39:09.619 INFO Finished test in 849239.0ms
> [MatterTest] 06-27 17:39:09.619 ERROR 
> 
> ******************************************************************
> *
> * Test test_TC_SOIL_2_2 failed for the following reason:
> * Details=SoilMoistureMeasuredValue is NullValue, Extras=None
> *
> * File "/home/grl/projectchip/masterconnectedhomeip/connectedhomeip/src/python_testing/TC_SOIL_2_2.py", line 89, in test_TC_SOIL_2_2
> * asserts.assert_true(measurement != NullValue, "SoilMoistureMeasuredValue is NullValue")

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

41c719200fc7796f8cebc1c631a46003859b1b19

### Platform

raspi

### Anything else?

**Sample app used to run the Test case:**
chip-all-clusters-app

**Reference Logs:**

[TC_SOIL_2_2.txt](https://github.com/user-attachments/files/20946363/TC_SOIL_2_2.txt)

**Python script References:**
-https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_SOIL_2_2.py

Ashwinigrl, 2025-07-09:
@ReneJosefsen

1.**TC-SOIL-2.2** - Test step 2  says `TH reads from the DUT the SoilMoistureMeasurementLimits attribute, save the MinMeasuredValue field as min_bound and save the MinMeasuredValue field as max_bound `. The MaxMeasuredValue field is incorrectly referenced as MinMeasuredValue second time.(Both min_bound and max_bound are assigned using MinMeasuredValue). Hence Test step label has to be updated correctly in the script. 

[image]

**Raised Test plan Issue for the same:** https://github.com/CHIP-Specifications/chip-test-plans/issues/5320

2. **TC-SOIL-2.1**

In  TC_SOIL_2_1, Step 11 is intended to verify that the value of SoilMoistureMeasuredValue is either null (i.e., NullValue) or a valid percent (uint8) between min_bound and max_bound.
 
However, the current script only checks the case when the value is not NullValue, and does not assert or log that a NullValue is also an acceptable and expected response.

[image]

**Could you please take a look at above issues also.**

Thanks!

ReneJosefsen, 2025-07-14:
Thanks for the additional points. I'm not sure how to address the second point, as far as I'm aware the script does not error on a null value, which is as intended? I could add a "if NullValue" but what would it assert at that point? So how I see the step it accepts null, or a non-null value within the min & max range?