project-chip/connectedhomeip #39794: [BUG] Crash with invalid pointer (this=0x100) in TLVWriter::StartContainer during TLV serialization

alessandro-mirri, 2025-07-01:
### Reproduction steps

I am using the Matter SDK (connectedhomeip) to create a C++ application that serializes and deserializes a light’s state (on/off) using the TLV system.

I implemented a simple LightState class with Serialize and Deserialize methods based on chip::TLV::TLVWriter and chip::TLV::TLVReader.

However, during runtime, the program crashes with a segmentation fault when entering this function:

```CHIP_ERROR TLVWriter::StartContainer(chip::TLV::Tag tag, chip::TLV::TLVType containerType, chip::TLV::TLVType & outerContainerType)```
From debugging with GDB, I found that the this pointer passed to StartContainer is invalid (0x100), causing an immediate crash.

What I did:
Properly initialize the buffer and call ```writer.Init(buffer, size);```

Verify with ```writer.IsInitialized()``` that the writer is valid right after Init()

Pass the writer by reference to the Serialize() method

Inside Serialize(), the call to StartContainer() crashes with this=0x100

Verified that the problem occurs only in StartContainer(), other methods like IsInitialized() work correctly when called before

Relevant GDB output:
```Breakpoint 1, chip::TLV::TLVWriter::StartContainer (this=0x100, tag=..., containerType=-10532, outerContainerType=<error reading variable: Cannot access memory at address 0x0>) at TLVWriter.cpp:549```
Additional info:
SDK version: release v1.4.2.0

OS: Ubuntu 24.04.2 LTS

Compiler: gcc 13.3.0

Minimal reproducible code snippet included below

Relevant code snippet (simplified):
```
uint8_t buffer[128];
chip::TLV::TLVWriter writer;
writer.Init(buffer, sizeof(buffer));
if (!writer.IsInitialized()) { /* handle error */ }
LightState light(true);
auto err = light.Serialize(writer);  // crash happens inside StartContainer
```

Questions
Could you help me understand if I’m missing something in the setup or use of TLVWriter?

Are there special precautions or initializations required before calling StartContainer?

Is it possible there is an ABI incompatibility between my build and the precompiled libraries?

Thank you very much for your help!

If you want, I can provide additional details such as my CMakeLists or a minimal reproducible project.

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

06523c22640ceb8b89f9a11ff2325a4481a178a3

### Platform

other

### Platform Version(s)

Linux Ubuntu 24.04.2

bzbarsky-apple, 2025-07-01:
@alessandro-mirri The code you have there looks correct. 

In your case, `0x100` is `this`, which is _really_ odd.  What does the code in `Serialize` look like?

But yes, if you are running this against pre-compiled binaries then ABI issues seem most likely here....

alessandro-mirri, 2025-07-02:
> [@alessandro-mirri](https://github.com/alessandro-mirri) The code you have there looks correct.
>
> In your case, `0x100` is `this`, which is _really_ odd. What does the code in `Serialize` look like? [...]

Sure, here it is.

#include "LightState.h"

using namespace chip;
using namespace chip::TLV;

CHIP_ERROR LightState::Serialize(TLVWriter &writer, Tag tag) const
{
    TLVType container;
    TLVType containerType = kTLVType_Structure;

    if (!writer.IsInitialized()) {
        std::cerr << "Writer non inizializzato correttamente\n";
        return CHIP_ERROR_INTERNAL;
    }

    ReturnErrorOnFailure(writer.StartContainer(tag, containerType, container));

    ReturnErrorOnFailure(writer.Put(ContextTag(1), isOn));
    return writer.EndContainer(container);
}

Program received signal SIGSEGV, Segmentation fault.
0x00005555555557de in chip::TLV::TLVWriter::IsInitialized (this=0x100) at /opt/connectedhomeip-1.4.2.0/src/lib/core/TLVWriter.h:1218
1218        bool IsInitialized() const { return mInitializationCookie == kExpectedInitializationCookie; }
(gdb) bt
#0  0x00005555555557de in chip::TLV::TLVWriter::IsInitialized (this=0x100) at /opt/connectedhomeip-1.4.2.0/src/lib/core/TLVWriter.h:1218
#1  0x000055555555c2a2 in chip::TLV::TLVWriter::StartContainer (this=0x100, tag=..., containerType=-10532, 
    outerContainerType=<error reading variable: Cannot access memory at address 0x0>) at ../../src/lib/core/TLVWriter.cpp:564
#2  0x0000555555555895 in LightState::Serialize (this=0x7fffffffd71e, writer=..., tag=...) at /home/merlino/workspace/light-matter/src/LightState.cpp:17
#3  0x00005555555554d6 in main () at /home/merlino/workspace/light-matter/src/main.cpp:25

bzbarsky-apple, 2025-07-02:
Callsite pasted above looks like this:

```
auto err = light.Serialize(writer);
```

Callee is defined like this:

```
CHIP_ERROR LightState::Serialize(TLVWriter &writer, Tag tag) const
```

how does that work?  Or is the tag defaulted to something?

Anyway, that gdb output shows that IsInitialized() is likewise getting a near-null pointer to the TLVWriter, so something is definitely odd here in terms of how things are getting passed...

alessandro-mirri, 2025-07-03:
> Callsite pasted above looks like this:
>
> ```
> auto err = light.Serialize(writer);
> ```
>
> Callee is defined like this:
>
> ```
> CHIP_ERROR LightState::Serialize(TLVWriter &writer, Tag tag) const
> ```
>
> how [...]

Yes, this works because it has a default parameter to TLV::AnonymousTag().
I don't know how to proceed. I also used valgrind and said me that there is a violation of memory when there is segmentation fault.

bzbarsky-apple, 2025-07-03:
I think at some point you need to start looking at the disassembly around the callsite and then the disassembly inside the callee and checking whether they agree on things like calling conventions (e.g. which register or stack location that TLVWriter argument is being passed in).