project-chip/connectedhomeip #39834: [BUG] Incorrect Write Privilege for Node Label attribute in Bridged Device Basic Information Cluster

dengpan-liu, 2025-07-03:
### Reproduction steps

The TC-ACE-2.2 fail for Matter Bridge with sub-devices:
```
Problem: ProblemSeverity.ERROR
    test_name: Write access checker - 3
    location:
        Endpoint: 3,
        Cluster:  57 (0x39) BridgedDeviceBasicInformation,
        Attribute:5 (0x05)
    problem: Unexpected error writing attribute - expected Unsupported Access, got 0
```
The root cause of this issue is the write privilege for Node Label attribute in Bridged Device Basic Information cluster  is not define in src/app/zap-templates/zcl/data-model/chip/bridged-device-basic-information.xml, but actually it shall only allow manage privilege for write operation.

### Bug prevalence

Only when TC-ACE-2.2 for Matter Bridge certification

### GitHub hash of the SDK that was being used

1dc089ce965e6b627b4c4ecd08b0e1882ece6b53

### Platform

other

bzbarsky-apple, 2025-07-08:
@dengpan-liu This was fixed in #38156.  It looks like you are off on a branch that predates that fix, so you may want to pick it up or backport it to the relevant branch.

closed 2025-07-08 by bzbarsky-apple

tongjiang, 2025-07-30:
@bzbarsky-apple 
This issue was also found in 1.4.0 branch, the Matter certificate team has report this issue in our Matter 1.4 matter bridge product.
Will you guys will cherry-pick the patch to this branch?
The pull #38156 has a lot of commit, so if I just want to fix this problem, how to pick the right commit in.
I found that the xml description is not the same as 1.4.0, so it's hard to pick the patch.

----in 1.4.0 the xml description is :
<attribute side="server" code="5"  define="NODE_LABEL"                type="char_string"               length="32"  default=""  writable="true" optional="true">NodeLabel</attribute>

----in the #38156 patch the description is :
    <attribute side="server" code="5" name="NodeLabel" define="NODE_LABEL" type="char_string" length="32" writable="true" optional="true">
      <access op="write" privilege="manage"/>
      <optionalConform/>
    </attribute>
the "NODE_LABEL" was moved to element "name".

bzbarsky-apple, 2025-07-31:
> and then make clean and recompile the whole code

Did you rerun codegen?

tongjiang, 2025-08-01:
> > and then make clean and recompile the whole code
> 
> Did you rerun codegen?

Yes