project-chip/connectedhomeip #39835: [Cameras] Refactor WebRTC Manager to handle streams in the transport and not the manager

marktrayer, 2025-07-03:
Per the comment from @pidarped in PR https://github.com/project-chip/connectedhomeip/pull/39779
"There is the mPeerConnection in the WebRTCTransport and the WebRTCTransportManager, redundantly. I think it should be only part of the WebRTCTransport so that the WebRTCTransportProviderManager can handle multiple sessions. The WebRTCSessionStruct is in the CurrentSessions attribute and, maybe, the WebRTCTransport can have a handle/ref on the corresponding WebRTCSessionStruct.

The above change for handling multiple clients and keeping a list of WebRTCTransports can potentially come in a separate PR. Please file an issue for that."

chulspro, 2025-07-04:
Okay, Thank you for raise issue. We will take care about this issue

robszewczyk, 2025-10-06:
@pidarped will take a look to check whether this is still a problem