project-chip/connectedhomeip #40216: [CERT-TEST-FAILURE] [TC-WEBRTC-1.5] Python script Failure

Ashwinigrl, 2025-07-21:
### Feature Area

Other

### Test Case

TC-WEBRTC-1.5

### Reproduction steps

1. **Command used to launch the DUT:** ./chip-camera-app
2. **Command used to run the python script :**  `python3 TC_WEBRTC_1_5.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json 
`
When we try to validate the TC-WEBRTC-1.5 python script, Test is failing at Test Step precondition-2 with the below error.

>   File "/home/ubuntu/master_jul/connectedhomeip/src/python_testing/TC_WEBRTC_1_5.py", line 106, in test_TC_WEBRTC_1_5
>     current_sessions = await self.read_single_attribute_check_success(
>                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/ubuntu/master_jul/connectedhomeip/no/lib/python3.12/site-packages/chip/testing/matter_testing.py", line 610, in read_single_attribute_check_success
>     asserts.assert_true(read_ok, read_err_msg)
>   File "/home/ubuntu/master_jul/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 319, in assert_true
>     fail(msg, extras)
>   File "/home/ubuntu/master_jul/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 441, in fail
>     raise signals.TestFailure(msg, extras)
> mobly.signals.TestFailure: Details=Error reading <class 'chip.clusters.Objects.WebRTCTransportRequestor'>:<class 'chip.clusters.Objects.WebRTCTransportRequestor.Attributes.CurrentSessions'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.UnsupportedCluster: 195>)), Extras=None
> [MatterTest] 07-21 09:07:55.499 INFO 		***** Test Failure : 
> [MatterTest] 07-21 09:07:55.500 INFO Finished test in 410101.0ms
> [MatterTest] 07-21 09:07:55.501 ERROR 
> 

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

252a32ba39f8497bf9406aab810bb4d8a77fd75c

### Platform

raspi

### Anything else?

**Reference Logs:**
[TC-WEBRTC-1.5.txt](https://github.com/user-attachments/files/21343801/TC-WEBRTC-1.5.txt)

**Python script References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_WEBRTC_1_5.py

marktrayer, 2025-09-03:
@Ashwinigrl WEBRTC_1_5 is currently disabled as it is missing support in the Camera App and other code areas.  So this will always fail and shouldn't be run.

@yufengwangca what's the status of supporting camera initiated flow?

yufengwangca, 2025-09-03:
This test case is disabled since it does not feasible from implementation point of view, we need clarification of its usage scenarios from spec side.
https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/12009