project-chip/connectedhomeip #40246: Consider making EventInfo() return failure when Cluster does not exist

Alami-Amine, 2025-07-22:
OK, now this could use a comment explaining why we need to ValidateClusterPath if we got EventInfo (the reason being that providers are allowed to lie and return something even if there is no such event)...  and maybe we should file a followup issue to require providers to return error if the cluster does not exist?

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/39772#discussion_r2220322694_

andy31415, 2025-07-31:
A caller can use codegen to try to get further info (e.g. get attributes and get a "invalid cluster" by listing Endpoints and ServerClusters) it is just incovenient for the caller (and slow if we actually have to consistently do it).

I think we have to describe the use case and how frequent it is. It is likely quite inexpensive for the Provider to allow for invalid endpoint/invalid cluster errors, it just means more code.

bzbarsky-apple, 2025-07-31:
> I think we have to describe the use case and how frequent it is

Right now every call in reporting engine has to double-check that the provider did not lie about the cluster instance existing.

bzbarsky-apple, 2025-08-01:
Yes, agreed.

soares-sergio, 2025-08-08:
When addressing this issue, please also audit/update here: https://github.com/project-chip/connectedhomeip/pull/39974/files#r2263671030