project-chip/connectedhomeip #40268: [BUG] ModeBase cluster have no GetInstance

ehsanziyaee, 2025-07-23:
### Reproduction steps

We can not use the public method of the Instance class cause GetInstance does not exist.

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

1.4

### Platform

esp32

### Platform Version(s)

1.4

### Anything else?

What are your suggestions for using the Mode-base cluster functions? We are overriding the delegate class in ASP.NET, but since the getinstance method is not available, we do not have access to the instance class.

bzbarsky-apple, 2025-07-23:
There is no single "instance" of this cluster (and often you will have multiple instances of it on different endpoints).  You're supposed to create instances of the cluster yourself, in your application, based on the endpoints you want it on.

bzbarsky-apple, 2025-07-28:
@ehsanziyaee So you are in a situation where you have a generic `ModeBase::Delegate`, not a specific subclass, and want to get the corresponding cluster instance?

ehsanziyaee, 2025-07-29:
@bzbarsky-apple yeah, we used esp-matter, we derived a class from ModeBase::Delegate in our code, but there is no 'GetInstance' function in protected section of ModeBase class, like other cluster, to access pulbic function of its Instance Class. all other cluster have this function.

bzbarsky-apple, 2025-07-30:
@ehsanziyaee if you're subclassing ModeBase::Delegate, then you can just use `mInstance`, no?  It's a protected member, not private.

ehsanziyaee, 2025-07-31:
@bzbarsky-apple Yeah, you’re right, but it’s better to have a getter/setter like the other clusters, isn’t it? 
Thanks for continuing to reply even after the issue was closed."

bzbarsky-apple, 2025-07-31:
Please feel free to open a PR that adds a getter.

closed 2025-09-22 by mergify[bot]