project-chip/connectedhomeip #40294: [BUG] WriteAttribute updates DefaultOTAProviders attribute despite ConstraintError in DefaultOTAProviders cluster

juandediosg, 2025-07-24:
### Summary
The `WriteAttribute` operation improperly updates the attribute value despite returning a `ConstraintError` status, causing inconsistent state in the `DefaultOTAProviders` attribute.

### Issue
During the `WriteAttribute` request on the `DefaultOTAProviders` attribute, the SDK returns a `ConstraintError (0x87) ` as expected for invalid writes, but the attribute value is still updated incorrectly. This causes the attribute to hold an invalid value despite the error response, causing inconsistent device behavior due the unexpected attribute update.

### RCA
It think this is an issue with the SDK implementation not using a shadow value. So it immediately changes storage as the list is processed, so even an aborted errored list write still leaves the list in a bad state

### Solution
Fix the cluster attribute write logic in `OtaSoftwareUpdateRequestorAttrAccess::WriteDefaultOtaProviders` to prevent value updates when a write operation fails with an error (e.g., `ConstraintError`). Also, add logging to trace which write paths are triggered in both success and failure scenarios.

### Related issues

- **Fixes:** [TC-SU-4.1](https://github.com/project-chip/matter-test-scripts/issues/659)
- **Matter specification:** [7.3.3 "Write Path Data Process](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#733-write-path-data-process) 
- **PR:** [TC-SU-4.1 python test #39980](https://github.com/project-chip/connectedhomeip/pull/39980)
- **PR updated with this verified steps for testing purpose** ([before](https://github.com/project-chip/connectedhomeip/blob/c5c49281071847d09ae1398bc538347c961a333e/src/python_testing/TC_SU_4_1.py#L339) , [after](https://github.com/project-chip/connectedhomeip/blob/c5c49281071847d09ae1398bc538347c961a333e/src/python_testing/TC_SU_4_1.py#L359) )

### Testing
During my implementation on  [TC-SU-4.1](https://github.com/project-chip/matter-test-scripts/issues/659), in Step 4, TH4 (on Fabric 1) sends a `WriteAttribute` request to update the `DefaultOTAProviders` attribute. The DUT correctly responds with a `ConstraintError (0x87)`, which is expected.

The TH4 write request attempted to set the attribute to:
`[OtaSoftwareUpdateRequestor.Structs.ProviderLocation(providerNodeID=4, endpoint=0, fabricIndex=1), OtaSoftwareUpdateRequestor.Structs.ProviderLocation(providerNodeID=1, endpoint=0, fabricIndex=1)]`

However, when TH4 read the attribute immediately before and after the failed write, I noticed the value was actually updated, even though the write was rejected.
- Before the write, the attribute showed:
  - `ProviderLocation(nodeID=1, endpoint=0, fabricIndex=1)`
- After the failed write, it changed to:
  - `ProviderLocation(nodeID=4, endpoint=0, fabricIndex=1)`

### Reproduce Test
To reproduce the issue, please follow the prerequisites and test steps described below.

#### Prerequisite: Build and run Matter Project
```bash
# Setup your Python environment
# Run Bootstrap
source scripts/bootstrap.sh
# Activate
source scripts/activate.sh
# Env
source out/python_env/bin/activate
```

#### Prerequisite: OTA

**Build  Components:**
```bash
# Build OTA Provider
scripts/examples/gn_build_example.sh examples/ota-provider-app/linux out/debug chip_config_network_layer_ble=false

# Build OTA Requestor
scripts/examples/gn_build_example.sh examples/ota-requestor-app/linux out/debug chip_config_network_layer_ble=false

# Build Chip Tool
scripts/examples/gn_build_example.sh examples/chip-tool out/chiptool 'chip_mdns="platform"'
  ```
**Create OTA Images: (these are just a stubs)**
```bash
# Create different firmware versions
echo "v1.0 firmware" > firmware_v1.bin
echo "v2.0 firmware" > firmware_v2.bin

# Generate OTA images
python3 src/app/ota_image_tool.py create -v 0xFFF1 -p 0x8001 -vn 1 -vs "1.0" -da sha256 firmware_v1.bin firmware_v1.ota
python3 src/app/ota_image_tool.py create -v 0xFFF1 -p 0x8001 -vn 2 -vs "2.0" -da sha256 firmware_v2.bin firmware_v2.ota

# Create some dirs for storage to avoid conflicts
mkdir /tmp/chip_tool_requestor
mkdir /tmp/chip_tool_provider 
```
#### Run Test

```bash
# Launch OTA Requestor (TH1 / DUT) from Terminal 1:
./out/debug/chip-ota-requestor-app --discriminator 1234 --passcode 20202021 --secured-device-port 5541 --autoApplyImage --KVS /tmp/chip_kvs_requestor 

# Run Python test with commission Provisioner/Requestor (Terminal 2):
python3 src/python_testing/TC_SU_4_1.py \
    --commissioning-method on-network \
    --discriminator 1234 \
    --passcode 20202021 \
    --vendor-id 65521 \
    --product-id 32769 \
    --nodeId 2
```

bzbarsky-apple, 2025-07-25:
This is expected/allowed behavior for list attributes.  Because list attribute values are not even guaranteed to be delivered in a single packet, and can be quite large, it's rather difficult, in general, to verify that the entire list is valid.

See Matter specification section "10.6.4.3.1. Lists" near the end:

> Because lists may be processed in chunks or by item, clients that receive an error status for a write action to a list attribute SHOULD NOT assume that the list contents are unchanged.

closed 2025-07-25 by bzbarsky-apple

tcarmelveilleux, 2025-07-25:
In the case of OTA requestor, if we mess-up the list, a device cannot update. I think if the case is "when you write > 1, only the first one remain" is what we shoudl check in the test. Perhaps this is a test change more than a code change.

bzbarsky-apple, 2025-07-25:
> In this case, however, the write is not partial or chunked

@juandediosg are you sure?  Because by default _all list writes get chunked_ (possibly within a single packet).  This issue does not include logs, so it's hard to say for sure, but I would fully expect that what the server receives is a "clear the list" followed by a sequence of appends, not a single "replace the list with this other list".

bzbarsky-apple, 2025-07-25:
> According to 7.3.3. Write Path Data Process when a write path returns an error status (such as ConstraintError), the attribute value shall not be changed:

Well, have you raised a spec issue about Data Model and Encoding contradicting each other here?

ivmarkov, 2025-08-29:
> > In this case, however, the write is not partial or chunked
> 
> [@juandediosg](https://github.com/juandediosg) are you sure? Because by default in the SDK _all list writes get chunked by the client_ (possibly within a single packet), with some exceptions recently added for ACL. This issue does not include logs, so it's hard to say for sure, but I would fully expect that what the server receives is a "clear the list" followed by a sequence of appends, not a single "replace the list with this other list". Crucially, at the point when it gets the first append, it has no idea when the "list write" will stop and whether that stop point will be "in the same packet" or not, so its only options are to process it immediately or to start buffering things up in RAM and hoping. And for constrained devices the latter is not a great option.

Sorry to come from nowhere and comment. I think this sheds some light on #40713 but maybe also amplifies that we need a more explicit language regarding data atomicity and consistency in the spec, as #40713 suggests?

juandediosg, 2025-09-01:
@ivmarkov Noted, thanks.