project-chip/connectedhomeip #40323: [TC-CLCTRL-4.1] The script can fail in step 3m if the DUT changes the latch value too slow

uebati-siegenia, 2025-07-28:
This is the test script issue to the test plan issue https://github.com/CHIP-Specifications/chip-test-plans/issues/5398

Test Case ID:
TC-CLCTRL-4.1

Describe Issue:
In test step 3j or in test step 3l the latch state changes to true. The time to let the device drive in this position is too short. We need to change this test step to a subscription.

Steps that require review:
Test step block 3.

If applicable: Test Event # this was found in:
Matter 1.5 TE#1

bzbarsky-apple, 2025-07-28:
@uebati-siegenia you may want to file this issue in https://github.com/project-chip/matter-test-scripts/issues because I think that's where test script issues are tracked, in the sense that people actually look at them and take action to fix them (though the actual test scripts live in this repository).

uebati-siegenia, 2025-07-29:
@bzbarsky-apple Hello, 
thank you to pointing me. I will reopen the issue at the right place.

uebati-siegenia, 2025-07-29:
Reopened on the correct place. https://github.com/project-chip/matter-test-scripts/issues/667

closed 2025-07-29 by uebati-siegenia