project-chip/connectedhomeip #40449: [CERT-TEST-FAILURE] Events Not Generated During Validation of Events Functionality Test Cases

Ashwinigrl, 2025-08-05:
### Feature Area

Other

### Test Case

 TC-DGSW-2.2  TC-SWTCH-3.2  TC-DGGEN-2.2

### Reproduction steps

1. Compile the sample app (All-cluster app) using the following command in the connectedhomeip folder:

`./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-arm64-all-clusters-no-ble-asan-clang build"`

2. Build the respective app (all-clusters-app) using the command:

`scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ examples/all-clusters-app/linux/out/all-clusters-app chip_inet_config_enable_ipv4=false`

3. Commission the DUT to the TH.

4. Open a second terminal on the DUT and obtain the PID of the DUT using:

`ps -aef | grep all-clusters-app`

5. Generate the event by following the steps below in the second terminal of the DUT:

A. Send the following command (replace <PID> with the PID obtained earlier):

`echo '{"Name":"SoftwareFault"}' > /tmp/chip_all_clusters_fifo_<PID>`

**Expected Outcome:**

The event should be generated on the first terminal of the DUT.

When the following command is executed on the Chip-tool, the generated event with all data should be successfully retrieved:
`./chip-tool softwarediagnostics read-event software-fault 1 0`

**Actual Outcome:**

Events are not generated on the first terminal of the DUT.

When the Chip-tool command is executed, the event with all data is not retrieved successfully.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

1b2b3fd571feaca7328db35798b99ab7ba4e6212

### Platform

raspi

### Anything else?

While validating the Events functionality test cases, the events are not being generated successfully, even after following the above steps.

**Sample app used to run the test:**
All-cluster-app

**The following test cases are affected:**

- [ ] TC-DGSW-2.2
- [ ] TC-SWTCH-3.2
- [ ] TC-DGGEN-2.2

**Test plan References:**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/switch.adoc#321-tc-swtch-32-primary-functionality-with-client-as-dut
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/software_diagnostics.adoc#tc-dgsw-2-2-event-functionality-dut_sever
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc#412-tc-dggen-22-event-functionality-dut_server

bzbarsky-apple, 2025-08-26:
@Ashwinigrl In your reproduction steps you don't say how you are actually running the DUT.  Did you pass it a `--app-pipe` argument?  What was the value of that argument?

Note that as far as I can tell, https://github.com/project-chip/connectedhomeip/pull/38778 changed how the pipe setup works significantly, and you seem to be doing things that are based on the situation from before that PR. But I can't tell for sure, since you didn't provide the infomation on how the application was run.

Ashwinigrl, 2025-08-28:
Thanks for input @bzbarsky-apple.  I ran the DUT with the --app-pipe argument, and it is working as expected. Hence closing this issue.

closed 2025-08-28 by Ashwinigrl