project-chip/connectedhomeip #40558: [CERT-TEST-FAILURE] [TC-CLCTRL-6.1]Need for Robust Event Matching: Handling Multiple Valid Events Before Target Event

bufl-blum, 2025-08-13:
### Feature Area

Other

### Test Case

TC-CLCTRL-6.1

### Reproduction steps

The test uses the _wait_for_event_report_ function in, for example, steps 5c and 5e to log the next event.

> This function allows a test script to block while waiting for a specific event to be the next event received, within a specified timeout (in seconds). It returns the event data so the values can be checked.

The problem is that if the SecureStateChanged event is transmitted before the MovementCompleted event, the test will fail.

A similar approach is used in steps 4b, 6c, and 6e, but there the TestEventTrigger function is used. In those cases, only one event is transmitted, so the test does not fail.

In test steps 7d, 7f, 8g, 8l, 9f, and 9l, the function _wait_for_secure_state_changed_and_assert_ is used to check for the SecureStateChanged event.

> This function waits for a SecureStateChanged event, allowing one other event to arrive first.

If more than one other event occurs before the SecureStateChanged event, the test step also fails.

My DUT briefly transmits both a "Disengaged" and an "Engaged" event before the SecureStateChanged event. The MovementCompleted event occurs last.

**Recommendation:**
We need a mechanism like await_all_expected_report_matches that can wait until all expected events occur (in any order), rather than only the next one or one out-of-order event. This would improve test robustness and prevent false negatives when multiple valid events are emitted in quick succession.

@AliTalebVelux please check thx

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

7b245457e2950177398765f28cc37f94dab1a0c2

### Platform

core

AliTalebVelux, 2025-08-13:
@bufl-blum: Thanks for created this issue. Please use this repo for test scripts issues: https://github.com/project-chip/matter-test-scripts

bufl-blum, 2025-08-13:
Reopened on the correct place: https://github.com/project-chip/matter-test-scripts/issues/675

closed 2025-08-13 by bufl-blum