project-chip/connectedhomeip #40599: [BUG] chip-tool pairs device, but java controller does not

oliviercailloux, 2025-08-14:
### Reproduction steps

I am trying to pair a commercial light bulb to get acquainted with the java example. Pairing works using `chip-tool` but not using an equivalent command with the `java-matter-controller`. Is this expected? Documentation suggests that it should work.

The light bulb is connected to the network already. Using its controller, I request a sharing matter code. Then running `out/host/chip-tool pairing code 1 xxxx-xxx-xxxx --paa-trust-store-path …/credentials/production/paa-root-certs` succeeds in pairing the light bulb. However, running `java -Djava.library.path=../lib/jni -jar java-matter-controller pairing code 1 xxxx-xxx-xxxx 10000 --paa-trust-store-path …/credentials/production/paa-root-certs` yields: `Error, Long discriminator is required` (and other errors). See attached log.

Linux (debian) host.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

91928f00ded3a4d29a524e186c9dfb0957dfee2d

### Platform

core

### Anything else?

[out.log](https://github.com/user-attachments/files/21779814/out.log)
Perhaps related to [this](https://github.com/project-chip/connectedhomeip/issues/24935)?

bzbarsky-apple, 2025-08-20:
The issue seems to be:

```
[1;31m[1755203063.319] [211597:211617] [SPT] VerifyOrDie failure at src/platform/CommissionableDataProvider.cpp:32: gCommissionableDataProvider != nullptr[0m
```

Why the java controller needs that, I have no idea.

s7092910, 2025-08-21:
From what I can tell, the Linux platform is setup as a BLE peripheral by default and chip-tool overrides the Linux platform to set up the BLE as a BLE central. The `java-matter-controller` example also doesn't override any of the platforms to set up the BLE as a BLE central. This causes the Linux `BLEManagerImpl` set up BLE advertising which is the cause of the crash. Only the python implementation of the controller forces the Linux `BLEManagerImpl` to act as the BLE central.

A work around until a fix can be made, I found adding the following to the `java-mater-controller` `CHIPProjectAppConfig.h` works.

```
#define CHIP_DEVICE_CONFIG_CHIPOBLE_ENABLE_ADVERTISING_AUTOSTART 0
```
It will prevent the BLE advertising from starting, preventing the crash.