project-chip/connectedhomeip #40675: [BUG] Matter Lightbulb discovered but fails to connect over ble-wifi

acolloredo, 2025-08-20:
### Reproduction steps

Run `sudo ./chip-tool pairing ble-wifi 233 <ssid> <pwd> 20491740 122` on Raspberry Pi 4 with Debian GNU/Linux v12

### Bug prevalence

>95% of the time on ssh, ~20% of the time directly on Pi

### GitHub hash of the SDK that was being used

c17bbcccad

### Platform

raspi

### Anything else?

I am attempting to commission a Matter Lightbulb using chip-tools ble-wifi pairing command. I am working off of a Matter build on a Raspberry Pi 4 that uses Bluez as the BLE connector. I had been working on a much older (~1 year old) version of Matter on commit 20200ee1df, so I recently reflashed the OS, recloned the repo, and rebuilt Matter. Since then, I have encountered this issue when I try to commission devices. Below are the logs of the chip-tool output from the following command:

`sudo ./chip-tool pairing ble-wifi 233 <ssid> <pwd> 20491740 122`
[bluez_failure.txt](https://github.com/user-attachments/files/21904339/bluez_failure.txt)

This occurs nearly every time I run this commission command when I am SSH'd into my Pi. When I connect my Pi to a monitor and keyboard and run the same command, it occurs much less frequently, although it occurs enough to still be an issue.

I have followed all the Bluetooth configuration steps in the Matter setup instructions, including enabling experimental features and disabling the battery plugin.

My Raspberry Pi, as well as my computer I SSH into, is connected to a wifi router in my home. #16178 suggests that this issue comes from an interference between WiFi and BLE on the Raspberry Pi, and to turn off wifi using `sudo nmcli radio wifi off` before commissioning. When I try this, the device is able to consistently connect via BLE but fails later on at `kFindOperationalForStayActive`. I have another Raspberry Pi with the same old Matter build (20200ee1df) flashed and do not encounter the same wifi-BLE interference issues. I have swapped SD cards between Pis to verify it is not a device issue either.

Thank you.

bzbarsky-apple, 2025-08-25:
@acolloredo My understanding is that in practice what you want to do is us a Bluetooth dongle, not the Pi's built-in Bluetooth support, if you use a Pi as a commissioner....  but I could be wrong here.

acolloredo, 2025-08-26:
Thank you very much @bzbarsky-apple! I will absolutely try that.