project-chip/connectedhomeip #40995: [CERT-TEST-FAILURE] [TC-PAVST-2.3] Python script failure

Ashwinigrl, 2025-09-15:
### Feature Area

Other

### Test Case

TC-PAVST-2.3

### Reproduction steps

1. When I am trying to validate the TC-PAVST-2.3 Python script, Test case is failing at step 9 **TH Reads ProvisionedEndpoints attribute from TLS Client Management Cluster on DUT**  due to the status code mismatch. It should be Success instead of ` "status" : "1 == kFailure"`.

> atterTest] 09-15 13:17:06.720 INFO 							"path" : 
> [MatterTest] 09-15 13:17:06.721 INFO 							{
> [MatterTest] 09-15 13:17:06.722 INFO 								"attribute_id" : "1 == 'provisionedEndpoints'",
> [MatterTest] 09-15 13:17:06.724 INFO 								"cluster_id" : "2050 == 'TlsClientManagement'",
> [MatterTest] 09-15 13:17:06.725 INFO 								"endpoint_id" : "1"
> [MatterTest] 09-15 13:17:06.726 INFO 							},
> [MatterTest] 09-15 13:17:06.727 INFO 							"status" : 
> [MatterTest] 09-15 13:17:06.728 INFO 							{
> [MatterTest] 09-15 13:17:06.729 INFO 								"status" : "1 == kFailure"
> [MatterTest] 09-15 13:17:06.730 INFO 							}
> [MatterTest] 09-15 13:17:06.731 INFO 						}
> [MatterTest] 09-15 13:17:06.732 INFO 					}
> [MatterTest] 09-15 13:17:06.734 INFO 				},
> [MatterTest] 09-15 13:17:06.735 INFO 				"interaction_model_revison" : "12",
> [MatterTest] 09-15 13:17:06.736 INFO 				"suppress_response" : "true"
> [MatterTest] 09-15 13:17:06.737 INFO 			}
> [MatterTest] 09-15 13:17:06.738 INFO 		},
> [MatterTest] 09-15 13:17:06.740 INFO 		"hex" : "15360115350037002402012503020824040118350124000118181818290424FF0C18",
> [MatterTest] 09-15 13:17:06.741 INFO 		"size" : 34
> [MatterTest] 09-15 13:17:06.742 INFO 	},
> [MatterTest] 09-15 13:17:06.743 INFO 	"payloadHeader" : 
> [MatterTest] 09-15 13:17:06.744 INFO 	{
> [MatterTest] 09-15 13:17:06.746 INFO 		"exchangeFlags" : 0,
> [MatterTest] 09-15 13:17:06.747 INFO 		"exchangeId" : 54841,
> [MatterTest] 09-15 13:17:06.748 INFO 		"initiator" : false,
> [MatterTest] 09-15 13:17:06.749 INFO 		"messageType" : 5,
> [MatterTest] 09-15 13:17:06.750 INFO 		"needsAck" : false,
> [MatterTest] 09-15 13:17:06.752 INFO 		"protocolId" : 1
> [MatterTest] 09-15 13:17:06.753 INFO 	}
> [MatterTest] 09-15 13:17:06.754 INFO }
> [MatterTest] 09-15 13:17:06.755 INFO >>> [E:54841i S:28388 M:9863050] (S) Msg RX from 1:0000000012344321 [FBFB] to 000000000001B669 --- Type 0001:05 (IM:ReportData) (B:64)
> [MatterTest] 09-15 13:17:06.785 ERROR Exception occurred in test_TC_PAVST_2_3.
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.12/dist-packages/mobly/base_test.py", line 801, in exec_one_test
>     test_method()
>   File "/usr/local/lib/python3.12/dist-packages/matter/testing/decorators.py", line 263, in async_runner
>     return _async_runner(body, self, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.12/dist-packages/matter/testing/decorators.py", line 251, in _async_runner
>     return test_instance.event_loop.run_until_complete(asyncio.wait_for(body(test_instance, *args, **kwargs), timeout=timeout))
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
>     return future.result()
>            ^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
>     return await fut
>            ^^^^^^^^^
>   File "/root/python_testing/scripts/sdk/TC_PAVST_2_3.py", line 215, in test_TC_PAVST_2_3
>     aProvisionedEndpoints = await self.read_single_attribute_check_success(
>                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.12/dist-packages/matter/testing/matter_testing.py", line 913, in read_single_attribute_check_success
>     asserts.assert_true(read_ok, read_err_msg)
>   File "/usr/local/lib/python3.12/dist-packages/mobly/asserts.py", line 319, in assert_true
>     fail(msg, extras)
>   File "/usr/local/lib/python3.12/dist-packages/mobly/asserts.py", line 441, in fail
>     raise signals.TestFailure(msg, extras)
> mobly.signals.TestFailure: Details=Error reading : = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError()), Extras=None
> [MatterTest] 09-15 13:17:06.897 ERROR 
> 
> ******************************************************************
> *
> * Test test_TC_PAVST_2_3 failed for the following reason:
> * Details=Error reading : = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError()), Extras=None
> *
> * File "/usr/local/lib/python3.12/dist-packages/matter/testing/matter_testing.py", line 913, in read_single_attribute_check_success
> * asserts.assert_true(read_ok, read_err_msg)
> *
> * Test step:
> *     9: TH Reads ProvisionedEndpoints attribute from TLS Client Management Cluster on DUT	Expected outcome: Store value as aProvisionedEndpoints.
> *
> * Endpoint: 1
> *
> *******************************************************************
> 
> [MatterTest] 09-15 13:17:06.898 INFO [Test] test_TC_PAVST_2_3 FAIL
> [MatterTest] 09-15 13:17:06.955 INFO Summary for test class TC_PAVST_2_3: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0
> [MatterTest] 09-15 13:17:06.960 INFO Summary for test run MatterTest@09-15-2025_13-16-44-350:
> Total time elapsed 22.559030559000007s
> Artifacts are saved in "/tmp/matter_testing/logs/MatterTest/09-15-2025_13-16-44-350"
> Test summary saved in "/tmp/matter_testing/logs/MatterTest/09-15-2025_13-16-44-350/test_summary.yaml"
> Test results: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0

2. In step 9 is referring the wrong PICS. It should be `TLSCLIENT.S.A0001`  instead of  `PAVST.S `.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

a82e43e06e35c707f9016c38ee83712c2ab58966

### Platform

raspi

### Anything else?

**Python script References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_PAVST_2_3.py

**Reference Logs:**

[TC-PAVST-2.3.log](https://github.com/user-attachments/files/22338094/TC-PAVST-2.3.log)

[image]

gmarcosb, 2025-09-15:
Will defer to @chulspro but can give some context:

`PAVST` is dependent on other clusters to actually work, specifically the TLS clusters to actually be able to do its work

But it is a test for `PAVST.S`, it simply needs some pre-setup (kind of like commissioning PICS is `OPCREDS.S` but every test must first do commissioning)

gmarcosb, 2025-09-15:
Per Push AV Stream Transport Cluster spec:

| Nodes supporting this cluster SHALL also support the TLS Client Management Cluster and its dependencies.

chulspro, 2025-09-17:
Thank you for updating. I will check and get back to you soon.