project-chip/connectedhomeip #41078: Support for Diagnostic Logs Transfer feature in Nordic DUT.

Saravana-kr22, 2025-09-22:
**Description:**

I’ve been testing diagnostic log retrieval using the test application and a binary provided by the Nordic team [merged.zip](https://github.com/user-attachments/files/22465722/merged.zip). The test app allows generating and retrieving logs stored under both user data and network data categories.

The workflow I followed:

1. Commission the device

```
./chip-tool generaldiagnostics test-event-trigger hex:00112233445566778899AABBCCDDEEFF 0xFFFFFFFF40000400 1 0  # save user data logs (1kB)  
./chip-tool generaldiagnostics test-event-trigger hex:00112233445566778899AABBCCDDEEFF 0xFFFFFFFF50000400 1 0  # save network data logs (1kB)  
```
2. Start interactive mode

```
./chip-tool interactive start 
 ```
3. Retrieve diagnostic logs
```
diagnosticlogs retrieve-logs-request 1 0 1 0  
diagnosticlogs retrieve-logs-request 0 0 1 0  
diagnosticlogs retrieve-logs-request 1 1 1 0 --TransferFileDesignator file_logs_1.txt  
diagnosticlogs retrieve-logs-request 0 1 1 0 --TransferFileDesignator file_logs_2.txt  
```
I was able to save logs into files (file_logs_1.txt, file_logs_2.txt).

Both user data and network data logs were retrieved successfully from the device.

**Observation / Gap:**

- The Diagnostic Logs Transfer cluster works correctly in the test app and binary.
- However, this feature is not yet implemented in the latest upstream connectedhomeip.
- Without upstream support, we cannot integrate this functionality into our Jenkins CI pipeline for automated testing and validation.

Request / Proposal:

- Add upstream support for the Diagnostic Logs Transfer feature in connectedhomeip.
- Ensure the cluster APIs (including retrieve-logs-request and TransferFileDesignator) are available in chip-tool and accessible for automation.
- This will allow us to:
  -    Automate log retrieval testing in CI pipelines.
  -    Validate device compliance with diagnostics log handling.
  -    Improve test coverage for diagnostic features beyond local test binaries.

bzbarsky-apple, 2025-10-15:
@kkasperczyk-no , I was told by @Damian-Nordic that you might be able to look at this.

kkasperczyk-no, 2025-10-16:
@bzbarsky-apple thanks for the heads up, I will check it with the team