project-chip/connectedhomeip #41126: [BUG] TC-CADMIN-1.19 Fails at Step 5b

Pradeep-grl, 2025-09-25:
### Reproduction steps

### Description:
The test case **TC-CADMIN-1.19** `fails` consistently at `step 5b` in both **TH GUI** and **Python execution inside Docker**.

### Failure Log from TH GUI:

> [MatterTest] 09-25 07:15:26.948 INFO Commissioning complete for node ID 0x0000000012344321: src/controller/CHIPDeviceController.cpp:1728: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
> [MatterTest] 09-25 07:15:26.949 WARNING Failed to commission: src/controller/CHIPDeviceController.cpp:1728: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
> [MatterTest] 09-25 07:15:26.949 ERROR Exception occurred in test_TC_CADMIN_1_19.
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.12/dist-packages/mobly/base_test.py", line 801, in exec_one_test
>     test_method()
>   File "/usr/local/lib/python3.12/dist-packages/matter/testing/decorators.py", line 263, in async_runner
>     return _async_runner(body, self, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.12/dist-packages/matter/testing/decorators.py", line 251, in _async_runner
>     return test_instance.event_loop.run_until_complete(asyncio.wait_for(body(test_instance, *args, **kwargs), timeout=timeout))
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
>     return future.result()
>            ^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
>     return await fut
>            ^^^^^^^^^
>   File "/root/python_testing/scripts/sdk/TC_CADMIN_1_19.py", line 119, in test_TC_CADMIN_1_19
>     await fids.CommissionOnNetwork(
>   File "/usr/local/lib/python3.12/dist-packages/matter/ChipDeviceCtrl.py", line 2949, in CommissionOnNetwork
>     return await asyncio.futures.wrap_future(ctx.future)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> matter.exceptions.ChipStackError: src/controller/CHIPDeviceController.cpp:1728: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
> [MatterTest] 09-25 07:15:26.956 ERROR 
> 
> * Test test_TC_CADMIN_1_19 failed for the following reason:
> * src/controller/CHIPDeviceController.cpp:1728: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
> 
> * File "/usr/local/lib/python3.12/dist-packages/matter/ChipDeviceCtrl.py", line 2949, in CommissionOnNetwork
> * matter.exceptions.ChipStackError: src/controller/CHIPDeviceController.cpp:1728: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
> 
> * Test step:
> *     5b: TH creates a controller on a new fabric and commissions DUT_CE using that controller	Expected outcome: Commissioning is successful
> * Endpoint: None
> 
> 

### Log Files:
**TH GUI Reference Logs:**
Controller Log: [TC-CADMIN-1_19_2025_09_25_12_45_14.log](https://github.com/user-attachments/files/22532002/TC-CADMIN-1_19_2025_09_25_12_45_14.log)
DUT Log: [TC-CADMIN-1_19-TH_DUT-log.txt](https://github.com/user-attachments/files/22532011/TC-CADMIN-1_19-TH_DUT-log.txt)

**Python Inside Docker Reference Logs:**
Controller Log: [TC-CADMIN-1_19-PythonInsideDocker-Failure-log.txt](https://github.com/user-attachments/files/22532038/TC-CADMIN-1_19-PythonInsideDocker-Failure-log.txt)
DUT Log: [TC-CADMIN-1_19-PythonInsideDocker-DUT-log.txt](https://github.com/user-attachments/files/22532042/TC-CADMIN-1_19-PythonInsideDocker-DUT-log.txt)

### Bug prevalence

whenever I do this.

### GitHub hash of the SDK that was being used

ac

### Platform

raspi

### Platform Version(s)

sac

### Anything else?

### Reference Screenshot: 

[image]

j-ororke, 2025-10-05:
Thank you for filing this issue and sharing the logs from your testing, Pradeep.  

I’ve been testing this case on my side and can confirm the behavior you’re seeing in the TH UI — step 5b still fails with the same error (`CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists`).  

```
******************************************************************
*
* Test test_TC_CADMIN_1_19 failed for the following reason:
* src/controller/CHIPDeviceController.cpp:1728: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
*
* File "/usr/local/lib/python3.12/dist-packages/matter/ChipDeviceCtrl.py", line 2949, in CommissionOnNetwork
* matter.exceptions.ChipStackError: src/controller/CHIPDeviceController.cpp:1728: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
*
* Test step:
*     5b: TH creates a controller on a new fabric and commissions DUT_CE using that controller	Expected outcome: Commissioning is successful
*
* Endpoint: 0
*
*******************************************************************

```

That said, when I run the same test via CLI inside the Docker container, it is currently passing for me.  

```
[MatterTest] 10-05 21:16:03.461 INFO Finished test in 68180550.0ms
[MatterTest] 10-05 21:16:03.463 INFO [Test] test_TC_CADMIN_1_19 PASS
```

Please note I’m still investigating the discrepancy between the TH UI and CLI runs and will update here as soon as I have a clearer root cause or workaround.

j-ororke, 2025-10-07:
Thank you, Cecille, 
You’re absolutely right. I had tried that earlier but initially ran into a timeout error, so I ended up troubleshooting in other directions. Your suggestion of incrementing the CA index by 1 during step 5b was indeed correct. I also needed to add the default_timeout property function to extend the test timeout from 3 minutes to 10 minutes, which appears to have resolved the issue and allows the test to pass in the TH UI here in my local test environment.

On my setup with minimal congestion it completes in about 4–5 minutes, but since this test fills the DUT’s fabric list and then removes it, I expect it could take closer to the full 10 minutes in a heavily congested lab environment. For now, I’ve set the timeout to 10 minutes to account for those cases.

TC_CADMIN_1_19 now passes locally in TH UI:
[image]