project-chip/connectedhomeip #41164: Please tell me how to get the current time.

NNAA0917, 2025-09-28:
I want to obtain the current time and save it as a log. How should I go about doing this?

What I am considering is using an RTC to get the latest time during commissioning, and then using a Timer Synchronization Cluster to perform time correction. Is this the correct approach?

NNAA0917, 2025-11-26:
Thank you.
I couldn't obtain the time from HomePod or commercial smart speakers.
Even though I forcibly set the device as the TrustedTimeSource, an "Access Deny" log appears.
I'm trying to get the time via SNTP, but it's not working.
Does anyone know how to implement SNTP on Thread while using the Matter SDK?

bzbarsky-apple, 2025-12-23:
You should implement Time Synchronization Server and let the commissioner tell you the time.

If you want to use SNTP, then you have to know an SNTP server you can reach and whatnot, which is not guaranteed (e.g. you might not be connected to the internet at all).